keshavdv/victron-ble #1: discover displays MAC address instead of id

ctschach, 2022-12-24:
Thanks for the work on this. I gave it a try, but running into a small issue….

When I run „discover“ it displays the MAC addresses of the device instead of the ble-id.
Running dump with the full MAC (including ‚:‘) works fine, running it without ‚:‘ does not display any data.
Therefore the „read“ command does not work properly. If you use the MAC address in the read (with ‚:‘) it failed because the invalid parameters, without ‚:‘ it does not deliver any data. Same if I use the ble-id.

`
oscar@calipi:~$ victron-ble -v discover
INFO:victron_ble.scanner:D7:70:BA:8E:59:A2: California Solar
INFO:victron_ble.scanner:C8:ED:93:A4:7B:7D: California Batterie
INFO:victron_ble.scanner:71:29:C9:E4:E0:B8: DumboVRM
INFO:victron_ble.scanner:FD:20:7C:A2:A9:8C: California BatSense
INFO:victron_ble.scanner:D1:DB:69:2F:5E:79: Dumbo SmartShunt
^C
Aborted!
oscar@calipi:~$ victron-ble dump FD:20:7C:A2:A9:8C
INFO:victron_ble.scanner:Dumping advertisements from FD:20:7C:A2:A9:8C
INFO:victron_ble.scanner:1671869618.4015198      : 1000a4a302bf72909828d92108a659ce376726090e2e23
^C
Aborted!
oscar@calipi:~$ victron-ble dump FD207CA2A98C
INFO:victron_ble.scanner:Dumping advertisements from FD207CA2A98C
^C
Aborted!

oscar@calipi:~$ victron-ble read "FD:20:7C:A2:A9:8C:906892428dfd869e50e65bc3544e2dfa"
Usage: victron-ble read [OPTIONS] [DEVICE_KEYS]...
Try 'victron-ble read --help' for help.

Error: Invalid value for '[DEVICE_KEYS]...': FD:20:7C:A2:A9:8C:906892428dfd869e50e65bc3544e2dfa is not a valid <addr>:<key> pair

oscar@calipi:~$ victron-ble read "'FD:20:7C:A2:A9:8C':906892428dfd869e50e65bc3544e2dfa"
Usage: victron-ble read [OPTIONS] [DEVICE_KEYS]...
Try 'victron-ble read --help' for help.

Error: Invalid value for '[DEVICE_KEYS]...': 'FD:20:7C:A2:A9:8C':906892428dfd869e50e65bc3544e2dfa is not a valid <addr>:<key> pair

oscar@calipi:~$ victron-ble read "FD207CA2A98C:906892428dfd869e50e65bc3544e2dfa"
INFO:victron_ble.scanner:Reading data for ['fd207ca2a98c']
^C
Aborted!

oscar@calipi:~$ victron-ble read "d6f1d6e7-34a4-c254-d013-c71e119513fa:906892428dfd869e50e65bc3544e2dfa"
INFO:victron_ble.scanner:Reading data for ['d6f1d6e7-34a4-c254-d013-c71e119513fa']
`

ctschach, 2022-12-24:
I'm working on a pull request....

keshavdv, 2022-12-24:
Ah, I've been testing on OSX where the mac address is hidden, I didn't realize the mac on other platforms includes the colon. The switch to a different delimiter in your PR makes a lot more sense now, but let's pick something besides the pipe.

closed 2022-12-24 by keshavdv

ctschach, 2022-12-24:
Maybe the easiest way would be to use two parameters instead of a delimiter between address and key

keshavdv, 2022-12-24:
I switched to the @ symbol in the latest release so hopefully that lets you get around this. I tried to keep it as one parameter so that you can pass multiple id@key pairs and it will display all of them in the same stream.

pioneersfx, 2024-05-23:
Hi, I opened an issue this morning and I think it comes down to a similar problem. The mac-address somehow produces an error. I tried multiple ways of formatting and nothing seems to work. Maybe someone has an idea :)

Here's my dumb from the terminal:

`victron-ble read 'f3:b8:a5:fd:b2:86@bdda8b53137339b45e9330be8c7bf70'
INFO:victron_ble.scanner:Reading data for ['f3:b8:a5:fd:b2:86']
ERROR:root:A message handler raised an exception: non-hexadecimal number found in fromhex() arg at position 31.
Traceback (most recent call last):
  File "src/dbus_fast/message_bus.py", line 811, in dbus_fast.message_bus.BaseMessageBus._process_message
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/manager.py", line 834, in _parse_msg
    self._run_advertisement_callbacks(
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/manager.py", line 878, in _run_advertisement_callbacks
    callback(device_path, device.copy())
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/scanner.py", line 279, in _handle_advertising_data
    self._callback(device, advertisement_data)
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/scanner.py", line 39, in _detection_callback
    self.callback(device, data)
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/scanner.py", line 107, in callback
    parsed = device.parse(raw_data)
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/devices/base.py", line 421, in parse
    decrypted = self.decrypt(data)
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/devices/base.py", line 405, in decrypt
    advertisement_key = bytes.fromhex(self.advertisement_key)
ValueError: non-hexadecimal number found in fromhex() arg at position 31

ERROR:root:A message handler raised an exception: non-hexadecimal number found in fromhex() arg at position 31.
Traceback (most recent call last):
  File "src/dbus_fast/message_bus.py", line 811, in dbus_fast.message_bus.BaseMessageBus._process_message
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/manager.py", line 834, in _parse_msg
    self._run_advertisement_callbacks(
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/manager.py", line 878, in _run_advertisement_callbacks
    callback(device_path, device.copy())
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/scanner.py", line 279, in _handle_advertising_data
    self._callback(device, advertisement_data)
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/scanner.py", line 39, in _detection_callback
    self.callback(device, data)
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/scanner.py", line 107, in callback
    parsed = device.parse(raw_data)
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/devices/base.py", line 421, in parse
    decrypted = self.decrypt(data)
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/devices/base.py", line 405, in decrypt
    advertisement_key = bytes.fromhex(self.advertisement_key)
ValueError: non-hexadecimal number found in fromhex() arg at position 31

ERROR:root:A message handler raised an exception: non-hexadecimal number found in fromhex() arg at position 31.
Traceback (most recent call last):
  File "src/dbus_fast/message_bus.py", line 811, in dbus_fast.message_bus.BaseMessageBus._process_message
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/manager.py", line 834, in _parse_msg
    self._run_advertisement_callbacks(
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/manager.py", line 878, in _run_advertisement_callbacks
    callback(device_path, device.copy())
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/scanner.py", line 279, in _handle_advertising_data
    self._callback(device, advertisement_data)
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/scanner.py", line 39, in _detection_callback
    self.callback(device, data)
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/scanner.py", line 107, in callback
    parsed = device.parse(raw_data)
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/devices/base.py", line 421, in parse
    decrypted = self.decrypt(data)
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/devices/base.py", line 405, in decrypt
    advertisement_key = bytes.fromhex(self.advertisement_key)
ValueError: non-hexadecimal number found in fromhex() arg at position 31

ERROR:root:A message handler raised an exception: non-hexadecimal number found in fromhex() arg at position 31.
Traceback (most recent call last):
  File "src/dbus_fast/message_bus.py", line 811, in dbus_fast.message_bus.BaseMessageBus._process_message
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/manager.py", line 834, in _parse_msg
    self._run_advertisement_callbacks(
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/manager.py", line 878, in _run_advertisement_callbacks
    callback(device_path, device.copy())
[... 142 lines cut ...]
    advertisement_key = bytes.fromhex(self.advertisement_key)
ValueError: non-hexadecimal number found in fromhex() arg at position 31

ERROR:root:A message handler raised an exception: non-hexadecimal number found in fromhex() arg at position 31.
Traceback (most recent call last):
[... cut ...]
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/scanner.py", line 107, in callback
    parsed = device.parse(raw_data)
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/devices/base.py", line 421, in parse
    decrypted = self.decrypt(data)
  File "/usr/local/lib/python3.9/dist-packages/victron_ble/devices/base.py", line 405, in decrypt
    advertisement_key = bytes.fromhex(self.advertisement_key)
ValueError: non-hexadecimal number found in fromhex() arg at position 31

^C
Aborted!
`

commit 35d7426f (2022-12-24): Switch to @ delimiter to fix #1