keshavdv/victron-ble #7: Can we add a "get_payload" to each parsed_data-object?

ctschach, 2022-12-29:
So I want to use the decoder library in my own application - using it in the documented way:
```py
from victron_ble.devices import detect_device_type

data = <ble advertisement data>
parser = detect_device_type(data)
parsed_data = parser(<key>).parse(<ble advertisement data>)
```

I do need to call every single value to construct an dict:
```py
result = {'temperature': parsed_data.get_temperature(), 'soc': parsed_data.get_soc() }      # and so on...
```

Wouldn't it be an option to add a "get_payload" to each of the DeviceData class which will deliver the complete payload with one call? 
So something like:
```py
class BatteryMonitorData(DeviceData):
    def __init__(self, data: Dict[str, Any]) -> None:
        self._data = data

    def get_payload(self):
        return self._data
```

keshavdv, 2022-12-29:
I avoided doing this intentionally at first to try and make the API more typing friendly and defer some of the legwork to consumers of the library. If this is critical, we could add explicit data types to encode the entire payload for each device and still maintain the type safety when adding something like this.

ctschach, 2022-12-29:
Okay, got your point...then let's leave this for later....

closed 2022-12-29 by ctschach