keshavdv/victron-ble #11: Add device type to decoded payload data?

ctschach, 2022-12-31:
What about adding the device type to the decoded data? Something like

`decoded_data.get_device_type()
`

Values could be generic descriptors like BMV, MPPT. or even more detailed names like "BatteryMonitor", "SolarCharger". I think this would be very helpful to know what functions to call to get the decoded data.

keshavdv, 2022-12-31:
Is e1b4c3cda6a5a745c8a1e0e70108c06a7ca986ff what you're looking for or do you want a more general device type lookup?

ctschach, 2023-01-01:
> Is [e1b4c3c](https://github.com/keshavdv/victron-ble/commit/e1b4c3cda6a5a745c8a1e0e70108c06a7ca986ff) what you're looking for or do you want a more general device type lookup?

More generic would make sense, but this will also work…

keshavdv, 2023-02-21:
I think the best way to do this is to use the types directly like in https://github.com/stefanor/signalk-victron-ble/blob/main/plugin.py#L51-L54 to decipher general device types. Closing for now unless you have some other ideas!

closed 2023-02-21 by keshavdv