keshavdv/victron-ble #27: Could not retrieve encryption keys using victronconnect on Windows 11

adrusa, 2023-05-15:
No advertisement keys in windows sqlite file

theHurc, 2023-05-17:
That is what I found as well, the sqlite file mentioned in the README is empty. You can pair a Victron controller to a Windows computer but the app will not connect to that paired device. The Windows VictronConnect app specifically says when clicking "Don't see the product you were looking for?" section "No Bluetooth on Windows"

I had to find a co-worker who had a Mac machine to get the keys

That begs the question: @keshavdv, why does the README for the repo suggest that the advertisement key is retrievable from a Windows machine? Is there something @adrusa and I are missing?

kaeser-sosae, 2023-05-21:
I'm running VictronConnect 5.9 on macOS and also find that the sqlite file's advertisementKeys table is empty. @theHurc, your colleague with the Mac, was he using an older version of VictronConnect do you know?

adrusa, 2023-05-23:
I finally used Linux Appimage on Ubuntu live usb to retrieve the keys and it worked well.

theHurc, 2023-05-24:
@kaeser-sosae, you still need to pair the Victron device with your Mac and connect to it in the app, that process will initiate the key transfer.

He was using version 5.90

joscelyn, 2023-09-05:
you can now access to advertisment keys in the victron application => settings => product information => encryption data

theHurc, 2023-09-11:
Yes! It works!!! Thanks for the update. This really makes commissioning a bunch of these devices easier.

BTW, the Encryption Data section in the app also contains the MAC address which is really handy.

3ricj, 2024-06-25:
> you can now access to advertisment keys in the victron application => settings => product information => encryption data

What application?  Android app?  Windows app?  Something else?

kaeser-sosae, 2024-06-25:
Yeah I don't see these keys in the MacOS app...