keshavdv/victron-ble #28: Node Red / MQTT / HA Support

ErikThiart, 2023-05-20:
Any plans to maybe create a stand alone / side project to implement this into Node Red or some other MQTT service?

I use https://solar-assistant.io/ it allows me to write data to it via MQTT.

Personally I wish they see this library and help you refine it and implementat it into their stack.

But realistically they won't but they do make provision for MQTT so if there's an easy to host on a PI or Mac app way to get this data and broadcast via MQTT to https://solar-assistant.io/ it would be epic.

rajlaud, 2023-05-26:
I have a working draft of a HomeAssistant plugin - see https://github.com/rajlaud/home-assistant/tree/victron-ble. Let me know if you want help setting it up as a custom component for testing purposes. I hope to get it submitted to HA soon. I've been testing it for a month or so now and it's worked really well.

rajlaud, 2023-07-07:
Draft HA pull request is here: https://github.com/home-assistant/core/pull/94994

They've asked that we move some of the code from the HA integration to this library; once those changes are merged in we should be ready to go.

rajlaud, 2024-03-22:
Just an update: I didn't have good luck getting my HA integration added to the HA core, but will try again soon. I'm waiting for a new release of this library with #47 and #48 merged; I'll update my integration with support for these and propose it again.

jedie, 2024-03-22:
I'm working on this, too. See: https://github.com/jedie/victron-ble2mqtt