keshavdv/victron-ble #31: CLI read function stops after reading in 11 messages

mflage, 2023-06-12:
**Describe the bug**
In the most recent version using the read CLI function only reads in the first 11 messages (albeit quite fast) and then stops reading. In the previous versions this worked by reading in a new message every x. second.

**To Reproduce**
Using the command victron-ble read <mac>@<serial> . I see 11 messages and then it stops and I have to restart the command to get new readings.

**Expected behavior**
I would expect the reading to continue indefinitely until the script is terminated with ^c.

**Desktop (please complete the following information):**
Raspberry Pi 4 running Raspbian Bullseye

**Additional context**
This worked in the previous version, but I can't remember which version that was. This was the pip package installed for Python 3.7 (which was the default one in Raspian Buster).

mflage, 2023-06-30:
Unfortunately I don't have different hardware/machine on this specific location. But I know for a fact that this worked just fine on the previous version of Raspian.

RomanDogdotCom, 2023-07-16:
I'm seeing this same issue.  After about a dozen run/kills ....app stops responding. "-- help" is only command to work. Will update as I find more info on issue.