keshavdv/victron-ble #38: reading out advertisment keys on linux (victron connect)

gkathan, 2023-08-29:
hi,
i am trying to read out the advertisment keys from my victron devices ... somehow the according database table is always empty EXCEPT one time where i suddenly had the data record pair (address+key) in my command line ... very strange because i cannot repeat that ... any ideas on that issue ?
greetings from vienna

joscelyn, 2023-09-04:
you can now access to advertisment keys in the victron application => settings => product information => encryption data

gkathan, 2023-09-04:
really ? awesome and thanx for providing the Infos ✨✨💪

gkathan, 2023-09-06:
hmmm on the android app I do not find the advertisment key 🤔
do I miss something?

joscelyn, 2023-09-12:
yep it's visible on my android app. 
maybe you should try to update it or install one on mac/windows if you can

gkathan, 2023-09-12:
hmmm ... not in my case ... I even updated manually to beta v6.0

screenshot attached (from the latest production release from september)

gkathan, 2023-09-12:
can you share a screenshot where the advertisement key is shown

joscelyn, 2023-09-12:
did you have enabled "Instand readout via Bluetooth" ?
here is a screenshot from my battery protect: https://i.ibb.co/kS421dN/Capture-d-e-cran-2023-09-12-a-10-58-40.png

gkathan, 2023-09-12:
as you can see on my screenshot i only have the option bluetooth = enabled
...
very strange ...

joscelyn, 2023-09-12:
i don't see your screenshot

gkathan, 2023-09-12:
> i don't see your screenshot

here we go ...[image]

gkathan, 2023-09-12:
😂😂wrong one 
.. this is it
[image]

joscelyn, 2023-09-12:
i don't see "Instand readout via Bluetooth" on you're screenshot, you should have both "Bluetooth" and "Instant readout via Bluetooth" because this project is taking advantage of instant readout (ble), for example my orion show both: https://i.ibb.co/VCKDGwX/Capture-d-e-cran-2023-09-12-a-16-25-56.png

joscelyn, 2023-09-12:
check on the internet if firmware and bootloader are indeed the latest version, otherwise maybe the revision of you're product doesn't handle ble :/