keshavdv/victron-ble #44: Porting to Micropython / Raspberry Pi Pico ?

petaramesh, 2024-01-31:
Hello,

I'm creating a hobby microcontroller project that involves reading Victron systems (a BMV-712 smart, a battery SmartSense, a Smart MPPT charger and an Orion DC-DC charger) on a Raspberry Pi Pico running Micropython.

I found out your project and it exactly fits the parts I miss for getting and processing the data from the Victrons.

I tested it on my Linux PC and it works like a charm.

So I tried installing the libraries onto my Pico, but bummer ! it doesn't work because Micropython doesn't have most of the standard CPython libraries you import - although it may generally provide what should be needed.

I'm not a seasoned Python dev myself - actually this is my first project in python, let alone on a microcontroller, but I have written all the code I need and it works, except for the Victron Bluetooth part that I'm still missing.

I don't feel at ease enough with your code for trying to port it myself to this environment, so i was wondering if it was in your own developments projects or if you could possiblly consider it ?

Many thanks in advance.

georg90, 2024-02-07:
HI @petaramesh would you mind sharing your code examples? I am in the same position as you and might be able to reuse / extend your work if you're okay with that. 

Thank you!

petaramesh, 2024-02-07:
Yeah I will try snipping the BT code from the rest of my project into something that can work standalone and post it here.
It will take me a little time.

georg90, 2024-02-07:
Very much appreciated

petaramesh, 2024-02-07:
Well here comes my quick and dirty extraction from my quick and dirty code.
If you have supported Victron devices and just put your devices MAC addresses and encryption keys in there and run it, it should start spitting received values to the console...

Make sure to install the asyncio librarires as indicated in the comments before attempting to run it.

I cut it from my whole project so it may still contain unnecessary code. I've tested it with a single Victron device I have on hand, and it works. So let me know...

```
# Pico_Victron_BT.py
#
# © Swâmi Petaramesh 2024
#
# This program is free software: you can redistribute it and/or modify it under the terms
# of the GNU General Public License as published by the Free Software Foundation,
# either version 3 of the License, or (at your option) any later version.
#
# This program is distributed in the hope that it will be useful, but WITHOUT ANY WARRANTY;
# without even the implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
# See the GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License along with this program.
# If not, see <https://www.gnu.org/licenses/>. 3
#
# This program is example code for receiving and decoding some Victron® devices information
# and statues over Bluetooth-LE on a Raspberry Pi® Pico W running :
#
# Works on : sys.version:3.4.0; MicroPython v1.22.1 on 2024-01-05
#            RPI_PICO_W-20240105-v1.22.1.uf2
#
# You need to put your own devices MAC addresses and encryption keys in the code below.

# Lbraries =============================================================
#
# Micropython built-in libraries
import time, struct, bluetooth, cryptolib
from machine import Pin
from cryptolib import aes

# Libraries to be installed on the Raspberry Pi Pico
# With mpremote tool (from Linux PC terminal) :
# mpremote mip install github:peterhinch/micropython-async/v3/primitives
# mpremote mip install github:peterhinch/micropython-async/v3/threadsafe
import asyncio
from threadsafe import ThreadSafeQueue

# Initialisations ==========================================================

_DEBUG = const(2)             # Extra debug console output

_UNKN = const(-9999)          # N/A value

pin_led_bt = Pin(25, Pin.OUT, value=1)         # Integrated LED, ON

# Victron devices parameters and values -----------------------------------

# We need to put our Victron devices MAC addresses and encryption keys here
# The rest should be left _UNKN

victron = {
    "bmv712" : { "mac": b"\xA3\x76\x6E\x7C\x22\x33",
            "key": b"\x12\x34\x56\x78\x9A\xBC\xDE\xF0\x12\x34\x56\x78\x9A\xBC\xDE\xF0",
            "volt": _UNKN,
            "amp": _UNKN,
            "soc": _UNKN,
            "temp": _UNKN,
            "upd": _UNKN
        },
    "orion" : { "mac": b"\xA3\x76\x6E\x7C\x22\x34",
           "key": b"\x12\x34\x56\x78\x9A\xBC\xDE\xF0\x12\x34\x56\x78\x9A\xBC\xDE\xF1",
           "eng_detect": True,
[... 394 lines cut ...]
    await asyncio.sleep(0)
    
    # Start Bluetooth BLE scanner
    scanner = BLEScanner(ble, maclist)
    scanner.start_scan(duration_ms=_BT_SCAN_DURATION_MS, interval_us=_BT_SCAN_INTERVAL_US, window_us=_BT_SCAN_WINDOW_US)
    
    while True:
        # Sets the main loop defined duration
        loop_begin_tick = time.ticks_ms()
        loop_end_target = time.ticks_add(loop_begin_tick,_TMR_MAIN_LOOP)
        
        # Wait until desired loop duration
        loop_time = time.ticks_diff(time.ticks_ms(),loop_begin_tick)
        last_time = comp_time + loop_time
        comp_time = 0
        loop_throttle = time.ticks_diff(loop_end_target,time.ticks_ms()) - 1
        if loop_throttle >= 0 :
            await asyncio.sleep_ms(loop_throttle)
        else:
            loop_throttle = 0

# Let's do it !
asyncio.run(main())

```

petaramesh, 2024-02-07:
I've changed some LED related values above, that's not that important, but it was wrong (as my original project uses LEDs in a different awy).

petaramesh, 2024-02-08:
You can these values in the VictronConnedct app on your phone. Go to your device, then “parameters”, menu, “Product info”, and down there “Show encryption key”.

georg90, 2024-02-08:
Thanks for pointing that out. I got it working with your instructions. 
See my adopted code here: https://gist.github.com/georg90/c6822fa28261059e4c8361bdcff13f32

I think this is a great basis for starting of a victron_ble port - maybe it can be considered. Otherwise anyone coming from Google will find this helpful. Thanks alot @petaramesh for the port!

petaramesh, 2024-04-07:
Hello,

I've updated my code to allow receiving and processing Victron BT advertisements even when the VictronConnect phone app is connected to the Victron device (previous code would only work when the VictronConnect app was not connected to the device).

So here is the new code :

```
# Pico_Victron_BT.py
#
# © Swâmi Petaramesh 2024
#
# This program is free software: you can redistribute it and/or modify it under the terms
# of the GNU General Public License as published by the Free Software Foundation,
# either version 3 of the License, or (at your option) any later version.
#
# This program is distributed in the hope that it will be useful, but WITHOUT ANY WARRANTY;
# without even the implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
# See the GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License along with this program.
# If not, see <https://www.gnu.org/licenses/>. 3
#
# This program is example code for receiving and decoding some Victron® devices information
# and statues over Bluetooth-LE on a Raspberry Pi® Pico W running :
#
# Works on : sys.version:3.4.0; MicroPython v1.22.1 on 2024-01-05
#            RPI_PICO_W-20240105-v1.22.1.uf2
#
# You need to put your own devices MAC addresses and encryption keys in the code below.

# Lbraries =============================================================
#
# Micropython built-in libraries
import sys, time, struct, bluetooth, cryptolib
from machine import Pin
from cryptolib import aes

# Libraries to be installed on the Raspberry Pi Pico
# With mpremote tool (from Linux PC terminal) :
# mpremote mip install github:peterhinch/micropython-async/v3/primitives
# mpremote mip install github:peterhinch/micropython-async/v3/threadsafe
import asyncio
from threadsafe import ThreadSafeQueue

# Initialisations ==========================================================

_DEBUG = const(2)             # Extra debug console output

_UNKN = const(-9999)          # N/A value

pin_led_bt = Pin("WL_GPIO0", Pin.OUT, value=1)         # Integrated LED, ON

# Victron devices parameters and values -----------------------------------

# We need to put our Victron devices MAC addresses and encryption keys here
# The rest should be left _UNKN

victron = {
    "bmv712" : { "mac": b"\xA3\x76\x6E\x7C\x22\x33",
            "key": b"\x12\x34\x56\x78\x9A\xBC\xDE\xF0\x12\x34\x56\x78\x9A\xBC\xDE\xF0",
            "volt": _UNKN,
            "amp": _UNKN,
            "soc": _UNKN,
            "temp": _UNKN,
            "upd": _UNKN
        },
    "orion" : { "mac": b"\xA3\x76\x6E\x7C\x22\x34",
           "key": b"\x12\x34\x56\x78\x9A\xBC\xDE\xF0\x12\x34\x56\x78\x9A\xBC\xDE\xF1",
           "eng_detect": True,
           "mode": _UNKN,
[... 396 lines cut ...]
    task_bt_decrypt = asyncio.create_task(bt_decrypt(bt_queue))                                # Decrypt and decode BT values
    await asyncio.sleep(0)
    
    # Start Bluetooth BLE scanner
    scanner = BLEScanner(ble, maclist)
    scanner.start_scan(duration_ms=_BT_SCAN_DURATION_MS, interval_us=_BT_SCAN_INTERVAL_US, window_us=_BT_SCAN_WINDOW_US)
    
    while True:
        # Sets the main loop defined duration
        loop_begin_tick = time.ticks_ms()
        loop_end_target = time.ticks_add(loop_begin_tick,_TMR_MAIN_LOOP)
        
        # Wait until desired loop duration
        loop_time = time.ticks_diff(time.ticks_ms(),loop_begin_tick)
        last_time = comp_time + loop_time
        comp_time = 0
        loop_throttle = time.ticks_diff(loop_end_target,time.ticks_ms()) - 1
        if loop_throttle >= 0 :
            await asyncio.sleep_ms(loop_throttle)
        else:
            loop_throttle = 0

# Let's do it !
asyncio.run(main())
```

jensimik, 2024-07-05:
https://github.com/jensimik/vicky
i made another micropython project with inspiration from the code here (on a tiny m5stick - around 20 euro)
intended to be used in a van with solar, dcdc and battery monitor.

joeyc64, 2024-07-11:
Maybe this is of some help?
https://github.com/joeyc64/victron_mqtt