keshavdv/victron-ble #45: Any Success with the BlueSmart chargers?

chicagoandy, 2024-02-12:
I would like to have my Blue Smart (IP67) chargers available in HomeAssistant.
https://www.victronenergy.com/chargers/blue-smart-ip65-charger

These are Victron "Smart" controllers, and they have Bluetooth available.   They are available in Victron Connect app via Bluetooth.

However, they do not use GAAT mode.   They do not have an advertisement key, or encryption key.    There is no encryption key in the app UI.   I have also inspected SQLITE db for Victron Connect and there is no Advertisement Key available there.

Can anyone point me in a direction to get these integrated in HomeAssistant.

Donixon, 2024-02-28:
Looking for the same implementation. Have the exact same Blue Smart chargers and we want to create a dashboard for them all how full the battery's are. Looking forward to hear if this is possible.

j9brown, 2024-03-01:
I have one of these devices too.  Unfortunately I don't think there's much we can do about them because they don't appear to publish their state via BLE like other devices.

chicagoandy, 2024-03-01:
They don't publish the realtime values via GATT, but they do publish the data.    Otherwise it wouldn't be available in Victron connect.
I would think it's a solvable problem.   Anyways, I've moved onto using MQTT instead of BLE, so it'll be for someone else to solve this one.

Donixon, 2024-03-13:
> They don't publish the realtime values via GATT, but they do publish the data. Otherwise it wouldn't be available in Victron connect. I would think it's a solvable problem. Anyways, I've moved onto using MQTT instead of BLE, so it'll be for someone else to solve this one.

How do you use MQTT? Do you have to WIFI version?

gregsyoung, 2024-05-01:
good news .. victron recently announced that “within next 3 months: a firmware update will enable BLE for the IP6X chargers. 
.. so hopefully after that we may see an update to support BLE reading.

j9brown, 2024-05-01:
Cool!  Let me know when it happens and I’ll implement support for it, assuming they publish the specs promptly.

gregsyoung, 2024-05-01:
see the message from victron rep in below discord thread 

https://discord.com/channels/1170433917761892493/1223726815248388208/1231019569490034698

gregsyoung, 2024-05-01:
i think there is a beta avail , u may be able to get early access to test

gregsyoung, 2024-05-25:
here is the victron msgs from the discord 
.. explains why as yet you cant update IP6x FW

[image]

gregsyoung, 2024-05-26:
[image]

Donixon, 2024-05-26:
> @Donixon 
> @j9brown 
> there is a signalk discord, the msg was in their from victron.
> (about 3 weeks back) 
> 
> it went something like …
> IP6X will be updated soon for BLE, its finalised and in testing, but could be “couple months” 
> and will be first released in victron connect app beta version.
> 
> i have downloaded latest beta of victron connect,  but like yourself , no joy as yet, no FW update showing for IP67s 
> 
> 
> 
> 

Nice! Funny I'm in the same discord for the free board pkuging in signal-k, but missed it completely. Hoping it will be here soon. Thanks for the share!

j9brown, 2024-05-26:
I’m looking forward to this update whenever it comes out. I’m busy for the
next few weeks so if it launches and anyone gets around to making the
necessary changes sooner please feel free to fork and send me a patch.

Donixon, 2024-10-10:
I saw that there was an update for the VictronConnect BETA app on: https://www.victronenergy.com/live/victronconnect:beta
```

 Phoenix & Blue Smart Chargers:

    Add firmware v3.61:
        Add Instant Readout
```

I'm not sure if that 'instant readout' is where the money is for us, but maybe worth a check. I don't have a BlueSmart Charger at hand right now, but maybe someone else?

librem5, 2024-11-28:
it works, https://github.com/keshavdv/victron-ble/pull/66
need add mapping for all bluesmart series
mapping here 
https://gist.github.com/seidler2547/52f3e91cbcbf2fa257ae79371bb78588

nopeee535, 2025-02-28:
are there any updates ? : )

j9brown, 2025-02-28:
Unfortunately my Blue Smart charger is too old and ineligible for the firmware update that adds instant readout.  I could perhaps blindly implement the code though if someone volunteers to test it.

nopeee535, 2025-02-28:
i would test it with my "Victron Energy Blue Smart IP67 12-Volt 25 Amp 230V"

j9brown, 2025-03-01:
It looks like Fabian has implemented the necessary bits in the esphome components for the Blue Smart chargers so that points the way towards what we need to do here.

https://github.com/Fabian-Schmidt/esphome-victron_ble/issues/62