keshavdv/victron-ble #46: Example for using this as library

hujiko, 2024-03-04:
Hey,

I tried using this library through the CLI and it worked fine.

Now as I would like to further process the information from my device, I would like to write my own python script and use your library as a library.

But sadly I don't really understand how to, based on the example given in the readme.
The readme says:

```
from victron_ble.devices import detect_device_type

data = <ble advertisement data>
parser = detect_device_type(data)
parsed_data = parser(<key>).parse(<ble advertisement data>)
``` 

But what's not clear to me, is: Where to get the BLE Advertisement Data from?
I tried using ID / MAC-Adress ("AA:BB:CC:DD:EE:FF") as ble advertisement data and the encryption key as "<key>", but apparently that did not work.

In the end, I would like to end up with the same JSON / Object, that I get from the CLI, so that I can persist that information in a database.

DannySkjodt1, 2024-03-09:
You can just mod the current binary for that, if you add quit() after the print line in line 114 it will quit after printing the json, you also have to comment out logger on line 73. Then you can call it from other stuff to grab your info.

TemperedEnterprises, 2024-03-30:
> You can just mod the current binary for that, if you add quit() after the print line in line 114 it will quit after printing the json, you also have to comment out logger on line 73. Then you can call it from other stuff to grab your info.

I did that and it still keeps running.. i put those changes in scanner.py

mystery-moss, 2024-10-18:
I wanted to do the same thing. I believe the readme assumes that you have acquired the Bluetooth advertisement data separately (e.g. with the Bleak library), and now want to parse it.

But if, like me, you don't care to learn how Bleak works, here's a hacky script to grab the data:

```
import asyncio
import json
from victron_ble.exceptions import AdvertisementKeyMissingError
from victron_ble.scanner import Scanner, DeviceDataEncoder

class ScannerWithRead(Scanner):
    """Scanner with the ability to read the latest received
    advertisement data to a variable.
    """
    def __init__(self, device_keys):
        super().__init__(device_keys)
        self.latest_data = None

    async def read(self):
        """Wait indefinitely until `callback` has populated
        `self.latest_data`
        """
        while self.latest_data is None:
            await asyncio.sleep(0.1)
        data = self.latest_data
        self.latest_data = None
        return data

    def callback(self, ble_device, raw_data):
        """Override `Scanner` callback to store data rather
        than printing it.
        """
        try:
            device = self.get_device(ble_device, raw_data)
        except AdvertisementKeyMissingError:
            return
        self.latest_data = device.parse(raw_data)

loop = asyncio.get_event_loop()

# Using MAC address and encryption key extracted as described in readme
scanner = ScannerWithRead({"AA:BB:CC:DD:EE:FF": "1234567891234567891234567891234"})
asyncio.ensure_future(scanner.start())

data = loop.run_until_complete(scanner.read())
data_dict = json.dumps(data, cls=DeviceDataEncoder)
print(data_dict)

loop.run_until_complete(scanner.stop())
```

Note that while this should work with v0.9.0, it is grubbing into the guts of victron_ble, so future updates may break it. I'm also not familiar with asyncio; this is likely not the ideal way to go about this, but it does work.

librem5, 2024-11-26:
Hello, you need use dump in bash or cmd:

victron-ble dump F9:64:15:03:D4:37
INFO:victron_ble.scanner:Dumping advertisements from F9:64:15:03:D4:37
INFO:victron_ble.scanner:1732580626.094929       : 100039a3087a0584a71115088e724e95a8f97d6719
INFO:victron_ble.scanner:1732580626.5600505      : 100039a3087b058426a604be99589c959d57c32693
INFO:victron_ble.scanner:1732580627.5106227      : 100039a3087c0584307ae3f28d1fc4c4d74c43b24e

where F9:64:15:03:D4:37 is your MAC-Adress

TemperedEnterprises, 2024-11-26:
If the documentation was superior to what it is, I might have made an open
source c# library

Why does everyone hate c#¿