keshavdv/victron-ble #52: Help with consuming as a library within script

IronDad75, 2024-03-24:
Hi, I’ve happily stumbled across Victron-ble whilst looking for a way to read VE stats in my leisure setup.

pretty much a noob with python so please be gentle :)….

I have installed Victron-ble on my raspberry pi4 and can run the read command through the rpi terminal command… (have pulled device mac and encryption key for devices)

Have replicated the terminal code (victron-ble read 1234@asdf)

really struggling to move this forwards and use with python script.

Any help would be really appreciated for an example code to read device data

thank you!

zefoo, 2024-05-12:
Hello! Did you ever get this? Having the exact same issue.

I can read fine via CLI and dump but cannot seem to get it with Python. Thank you!

IronDad75, 2024-05-12:
Hi Zefoo, yes I did! Got some help with this over on stackoverflow. Here’s a link. 
https://stackoverflow.com/questions/78220963/constructing-a-script-using-a-3rd-party-library/78227258#78227258

took it a stage further to write data out to a txt file and stopped the scan running once file was written. 👍🏻

keshavdv, 2024-10-09:
Glad you got this sorted, going to close this one out for now!

closed 2024-10-09 by keshavdv