keshavdv/victron-ble #60: Add support for Blue Smart IP67 Charger Waterproof

Donixon, 2024-10-15:
I saw that there was an update for instant read-out on the Blue Smart Chargers. Tried this with Victron-BLE with no luck. It didn't recognize the charger. The chargers we got are the: https://www.victronenergy.com/chargers/blue-smart-ip67-charger-waterproof 
Here the asked dump:

**victron-ble discover**
`INFO:victron_ble.scanner:C0:68:7B:78:58:04: None`

**victron-ble read**
`INFO:victron_ble.scanner:Reading data for ['c0:68:7b:78:58:04']
ERROR:victron_ble.scanner:Could not identify device type for C0:68:7B:78:58:04: None
ERROR:victron_ble.scanner:Could not identify device type for C0:68:7B:78:58:04: BSC IP67 24/8 HQ2248ZAF2F`

**victron-ble dump**
`INFO:victron_ble.scanner:Dumping advertisements from C0:68:7B:78:58:04
INFO:victron_ble.scanner:1728982681.1049268      : 10001ca3087d07a80e68ee71b979b8e2645299435d
INFO:victron_ble.scanner:1728982682.0453758      : 10001ca3087e07a85578b9f6caef91cab596cf35b7
INFO:victron_ble.scanner:1728982683.621407       : 10001ca3087f07a853e2af27dbfb4bdb2b94080a17
INFO:victron_ble.scanner:1728982684.4092534      : 10001ca3088007a83b30291d79be40f5c1347bca49
INFO:victron_ble.scanner:1728982686.2969787      : 10001ca3088207a86b961a41094653f976b6dbaac5
INFO:victron_ble.scanner:1728982690.6899915      : 10001ca3088707a8e0efeac8ba18983a126a112334
INFO:victron_ble.scanner:1728982691.6332154      : 10001ca3088807a845da57eaab721eb25a61ec9277
INFO:victron_ble.scanner:1728982698.117435       : 10001ca3088e07a83eb93968716f127e3c4b1c89b0
INFO:victron_ble.scanner:1728982699.0635798      : 10001ca3088f07a8e2f62dba8f0541f0981f866706
INFO:victron_ble.scanner:1728982700.4813232      : 10001ca3089007a814eb16cb2be830c316feba1a6f
INFO:victron_ble.scanner:1728982700.9471943      : 10001ca3089107a804c08454420b86495911a338bb
INFO:victron_ble.scanner:1728982702.3676858      : 10001ca3089207a83be5e8d4e3409b16b43bc3215a
INFO:victron_ble.scanner:1728982703.4661026      : 10001ca3089307a881dcfa6bf8a44e92b9dde9c38a
INFO:victron_ble.scanner:1728982703.785097       : 10001ca3089407a8c9f1cc08830498558748530ea0
INFO:victron_ble.scanner:1728982704.7453043      : 10001ca3089507a833fd01e06b8ef9203e59814a57
INFO:victron_ble.scanner:1728982705.68234        : 10001ca3089607a8888ee77bc942b6c1cab8b50469
INFO:victron_ble.scanner:1728982708.047632       : 10001ca3089807a8d9f25055320df2a4c290ee3632
INFO:victron_ble.scanner:1728982709.131191       : 10001ca3089907a8c06b7a3a5df0b5a25fbeeaeafe
INFO:victron_ble.scanner:1728982710.0623465      : 10001ca3089a07a8245c196dadf03599e7773e2ad1
INFO:victron_ble.scanner:1728982711.0049634      : 10001ca3089b07a8c3b5115e0522860d1d822192df
INFO:victron_ble.scanner:1728982711.9472325      : 10001ca3089c07a80b06d75d93fc94cf276f3859fa
INFO:victron_ble.scanner:1728982713.353701       : 10001ca3089d07a814bb675a4528186a79c4225b35
INFO:victron_ble.scanner:1728982713.8279777      : 10001ca3089e07a8131be4de13adfc461cbdc14736
INFO:victron_ble.scanner:1728982714.7755136      : 10001ca3089f07a8ae470c90d486d371cb0747f96d
INFO:victron_ble.scanner:1728982716.6578944      : 10001ca308a107a8057dd7473c12cf85018b8c10db
INFO:victron_ble.scanner:1728982718.0853612      : 10001ca308a207a8a416c403789dd4a0c1fd034f75
INFO:victron_ble.scanner:1728982719.0352182      : 10001ca308a307a8ac261430e39ff29d5cbcc3a958
INFO:victron_ble.scanner:1728982719.9700894      : 10001ca308a407a81d005e2a0191aed997198323d6
INFO:victron_ble.scanner:1728982720.9123447      : 10001ca308a507a8db6bbb22f68d0423e5e172651f
INFO:victron_ble.scanner:1728982721.867222       : 10001ca308a607a8725775b8080dd2f197fb040105
INFO:victron_ble.scanner:1728982722.815946       : 10001ca308a707a8889de021d3eb90dbb5b2e49d96
INFO:victron_ble.scanner:1728982723.7539349      : 10001ca308a807a88d9c636769051ed9bcc3841496
INFO:victron_ble.scanner:1728982724.7083602      : 10001ca308a907a8af03e7842867dba85836e1544f
INFO:victron_ble.scanner:1728982725.6389837      : 10001ca308aa07a8d9e7c2a225873954c52d4fb372
INFO:victron_ble.scanner:1728982727.6743166      : 10001ca308ac07a85da38807124ebf3a0b8beebe7a
INFO:victron_ble.scanner:1728982728.9453428      : 10001ca308ad07a8621d7181e7d3772b9670aadcd4
INFO:victron_ble.scanner:1728982731.9064507      : 10001ca308b007a8a5497b696a815663e81458db43
INFO:victron_ble.scanner:1728982733.3381417      : 10001ca308b107a8d7e773ffeba38e71fb2d1fd185
INFO:victron_ble.scanner:1728982738.0539434      : 10001ca308b607a863efc6c04feac76a93d557a95e
INFO:victron_ble.scanner:1728982738.6713476      : 10001ca308b707a89334a42d5f077d83cdec0c1fb2
INFO:victron_ble.scanner:1728982739.946365       : 10001ca308b807a8a7d3b1601310cd596acdde66c6
INFO:victron_ble.scanner:1728982740.8820665      : 10001ca308b907a8d9c0d9fe4bf5325a8146058f51
INFO:victron_ble.scanner:1728982742.2966678      : 10001ca308ba07a80cd8ddc970a6090e1a77f519bd
INFO:victron_ble.scanner:1728982745.2816558      : 10001ca308bd07a85b9db6ada05323bca517cc97e6
INFO:victron_ble.scanner:1728982746.5482712      : 10001ca308be07a8d2a1d283eb1c869eb993de8181
INFO:victron_ble.scanner:1728982749.9788299      : 10001ca308c207a8b9fbf7d8d40d2e02c0c500f655
INFO:victron_ble.scanner:1728982751.3936446      : 10001ca308c307a884a4a68abf61034a61cfe0efe4
INFO:victron_ble.scanner:1728982751.8715427      : 10001ca308c407a87da7a1a126cb0cdcac29e57a02
INFO:victron_ble.scanner:1728982754.7106626      : 10001ca308c707a84470605145ea93572970934b5c
INFO:victron_ble.scanner:1728982756.597156       : 10001ca308c807a80795e8c49bd8a6de5155b31c11
INFO:victron_ble.scanner:1728982757.543178       : 10001ca308c907a82fa9574aee62bbc60254aefc23
INFO:victron_ble.scanner:1728982760.966231       : 10001ca308cd07a857860abfb2997fb80bf802c481
INFO:victron_ble.scanner:1728982763.6730757      : 10001ca308d007a888bef0d8aeb8520f868748be2c
INFO:victron_ble.scanner:1728982765.234188       : 10001ca308d107a815602a4fc16a5352b3990bc998
INFO:victron_ble.scanner:1728982766.638106       : 10001ca308d307a834c010804d536ff75c9fb12b5d
INFO:victron_ble.scanner:1728982768.985559       : 10001ca308d507a885116a16c16904789137eba35f
INFO:victron_ble.scanner:1728982769.9369721      : 10001ca308d607a8ce33e0780ed890d4e5d456ea8e
INFO:victron_ble.scanner:1728982771.3490617      : 10001ca308d707a87ec25ec62d10e4ea849fb9863f
INFO:victron_ble.scanner:1728982773.3704083      : 10001ca308d907a898c4d1f3d7e8bbf5d6b8b531ec
INFO:victron_ble.scanner:1728982774.7879627      : 10001ca308db07a8b8d0f7bf1d868702f207d73815
INFO:victron_ble.scanner:1728982776.6779537      : 10001ca308dd07a80f1e96c575f37183ca55306448
INFO:victron_ble.scanner:1728982778.088123       : 10001ca308de07a8b818758ae24153c26c83741a71
INFO:victron_ble.scanner:1728982779.0258055      : 10001ca308df07a8e1604df646b459f633eef3afee
INFO:victron_ble.scanner:1728982779.9726198      : 10001ca308e007a8c60f9defc7c0635a2904213712
INFO:victron_ble.scanner:1728982780.9183962      : 10001ca308e107a883a88da964ec5bd0cf1ee94ce2
INFO:victron_ble.scanner:1728982781.8704517      : 10001ca308e207a8954a7f666ad44c6d64c945ec9a
INFO:victron_ble.scanner:1728982783.761472       : 10001ca308e407a8580a8b5d4ac8ed532dcae05fc3
INFO:victron_ble.scanner:1728982784.7031112      : 10001ca308e507a835e54e463b5684dbbf209d7033
INFO:victron_ble.scanner:1728982785.6412466      : 10001ca308e607a826c79c48f6c1b82e18430fefd6
INFO:victron_ble.scanner:1728982787.0712442      : 10001ca308e707a8936e5f403b4fd7bef9701780e0
INFO:victron_ble.scanner:1728982788.0218456      : 10001ca308e807a8d9ab8e4c85ea6297e647545948
INFO:victron_ble.scanner:1728982790.5064774      : 10001ca308ea07a8aa1143a09a42e97750ad99e075
INFO:victron_ble.scanner:1728982791.9258258      : 10001ca308ec07a8620261a467546446e1c5f558b0
INFO:victron_ble.scanner:1728982792.8577652      : 10001ca308ed07a80fcf60d3f13fecb26afb73a4c2`

I hope this is the correct way to do this. If there is any other info needed, i'm glad to give it.

Donixon, 2024-10-15:
And the screenshots:
[image]

[image]

[image]

Donixon, 2024-10-16:
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/KRVEDO~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/KRVEDO~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#467886" vlink="#96607D">

Cycle # | Bulk time (m) | Absorption time (m) | Recondition time (m) | Float time (m) | Storage time (m) | Bulk charge (Ah) | Absorption charge (Ah) | Recondition charge (Ah) | Float charge (Ah) | Storage charge (Ah) | Start voltage (V) | End voltage (V) | Battery Type | Termination reason | Error
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
0 | 0 | 30 | 0 | 30 | 0 | 0.0 | 3.7 | 0.0 | 3.8 | 0.0 | 28.79 | 28.80 | normal | Completed | #0 No error

</body>

</html>

Also could export a cycle-history.csv. Included like above. Maybe its usefull.

librem5, 2024-11-26:
Victron ProductId to product mapping (velib products.h)
https://gist.github.com/seidler2547/52f3e91cbcbf2fa257ae79371bb78588
mapping  for blue chargers ip65
41784 Blue Smart IP65 Charger 12/10
41785 Blue Smart IP65 Charger 12/15
41786 Blue Smart IP65 Charger 24/5
41787 Blue Smart IP65 Charger 24/8
41788 Blue Smart IP65 Charger 12/5

librem5, 2024-11-26:
41785 Blue Smart IP65 Charger 12/15
INFO:victron_ble.scanner:F9:64:15:03:D4:37: None
victron-ble dump F9:64:15:03:D4:37
[dump.txt](https://github.com/user-attachments/files/17910690/dump.txt)

closed 2024-11-26 by Donixon

librem5, 2024-11-28:
Finally it working. at my Blue Smart IP65 Charger 12|15
Next Commit i will add, all smart charges series
https://github.com/keshavdv/victron-ble/pull/66