keshavdv/victron-ble #61: Home Assistant HACS getting error from victron-ble

kencraw, 2024-10-16:
**Describe the bug**
Home Assistant is not able to read the output of my SmartSolar MPPT controller.  Installation of HACS and Victron-BLE went fine.  HA discovers the SmartSolar device automatically.  But when it shows up in the list of devices as "SmartSolar HQ22394J2HQ" the 2nd line says "No devices or entities"

The one thing I *may* have misconfigured is the "Advertising Key" as there seems to be a terminlogy inconsistency.  Using my Android VictronConnect app I followed the directions on the main page (https://github.com/keshavdv/victron-ble) and got to the "Instant Readout Encryption" screen.  I checked that the Mac Address matched with the discovered device in HA was showing and then typed in the provided "Encryption Key" into the "Advertisement Key" on HA.  It seems like that it what I should do, but since the terms don't match I wanted to make clear this is how I configured it.

**Steps to reproduce:**
1. Install HACS and Victron-BLE on HA
2. Go to HA -> Settings -> 'Devices & Services'
3. In the 'Discovered' section, click the 'Configure' button
4. Review the name and address in the Pop-Up window match the device
5. Enter the "Encryption Key" from the Android app (as described above) into the "Advertisement Key" box.
6. Click Submit
7. Click Finish on box indicating successful setup
8. In HA enable Debug Logging for Victron BLE
9. Go to HA -> Settings -> System -> Logs
10. Click on "Unexpected error updagting SmartSolar HQ22394J2HQ data" entry under "Home Assistant Core" section
11. Copy error text

**Log details of error:**
Here is the Error in the HA Log:

This error originated from a custom integration.

Logger: custom_components.victron_ble
Source: components/bluetooth/passive_update_processor.py:371
integration: Victron BLE (documentation, issues)
First occurred: October 15, 2024 at 9:18:47 PM (30825 occurrences)
Last logged: 5:52:44 AM

Unexpected error updating SmartSolar HQ22394J2HQ data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/passive_update_processor.py", line 371, in _async_handle_bluetooth_event
    update = self._update_method(service_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bluetooth_sensor_state_data/__init__.py", line 70, in update
    self._start_update(data)
  File "/config/custom_components/victron_ble/device.py", line 58, in _start_update
    self._process_mfr_data(address, local_name, mfr_id, mfr_data, service_uuids)
  File "/config/custom_components/victron_ble/device.py", line 73, in _process_mfr_data
    parsed = device_parser(self.key).parse(data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/victron_ble/devices/base.py", line 444, in parse
    decrypted = self.decrypt(data)
                ^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/victron_ble/devices/base.py", line 428, in decrypt
    advertisement_key = bytes.fromhex(self.advertisement_key)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: non-hexadecimal number found in fromhex() arg at position 31
(end error)

**Versions:**
Home Assistant: 2024.10.2 on Operating System 13.2
Victron-BLE: 0.1.1
HW: Raspberry Pi 4 (Using stock RPi image of HA)

Thanks in advance for any support you can provide.

Jan-789, 2024-11-28:
Same error : ValueError: non-hexadecimal number found in fromhex() arg at position 31
versions:
Home assistant 2024.11.3 on OS 13.2
Victron BLE 0.1.1 Modified integration
HW Raspberry Pi 4b

Jan-789, 2024-12-19:
Turned out that the advertisement key was 1 position short, regenerate the advertisement key by changing the pin-number fixed the problem.