keshavdv/victron-ble #67: Add support for IP6x chargers

KvotheBloodless, 2025-01-03:
**Is your feature request related to a problem? Please describe.**
I would like to be able to collect data from Victron IP6x chargers.

**Describe the solution you'd like**
Victron [recently added](https://community.victronenergy.com/t/blue-smart-charger-firmware/7233/2) instant readout support to their range of IP6x chargers. It would be great to see support added to the library.

**Describe alternatives you've considered**
None

**Additional context**
Related ticket open [here](https://github.com/stefanor/signalk-victron-ble/issues/19).

Shane11r, 2025-01-06:
INFO:victron_ble.scanner:E7:94:5B:3D:F4:53: None

INFO:victron_ble.scanner:Reading data for ['e7:94:5b:3d:f4:53']
ERROR:victron_ble.scanner:Could not identify device type for E7:94:5B:3D:F4:53: None
ERROR:victron_ble.scanner:Could not identify device type for E7:94:5B:3D:F4:53: BSC IP65 12/10 HQ2341NQ7PG

INFO:victron_ble.scanner:Dumping advertisements from e7:94:5b:3d:f4:53
INFO:victron_ble.scanner:1736142929.7465699      : 100038a308d20ff61d538d85f4dbf4f0d5fe22af5e
INFO:victron_ble.scanner:1736142930.6876442      : 100038a308d30ff61a47cc7775aac97ac04ab28c0b
INFO:victron_ble.scanner:1736142931.6273985      : 100038a308d40ff668e85eb7403ca40f3c8ad9afed
INFO:victron_ble.scanner:1736142936.4978874      : 100038a308d80ff62987ed7d3c85fcb5dda390c75b
INFO:victron_ble.scanner:1736142937.7681704      : 100038a308da0ff6db99e6e0c395cb29bb41e1edbf
INFO:victron_ble.scanner:1736142938.7263515      : 100038a308db0ff6df7a1a26abf9b38622206643fc
INFO:victron_ble.scanner:1736142940.6085937      : 100038a308dd0ff6221e2423f99e140935b68c036d
INFO:victron_ble.scanner:1736142942.496592       : 100038a308de0ff6afd670b8fc93b6dca475c308d5
INFO:victron_ble.scanner:1736142943.4436364      : 100038a308df0ff600db87905bd7a222788230b3fb
INFO:victron_ble.scanner:1736142943.9173486      : 100038a308e00ff683284a61154d3dea5c356d88b5
INFO:victron_ble.scanner:1736142945.3314855      : 100038a308e10ff6f475e0c5938fc7508d7e6dd07b
INFO:victron_ble.scanner:1736142948.1672707      : 100038a308e40ff62d3da605c5697ccb63f974eb6c
INFO:victron_ble.scanner:1736142949.1165993      : 100038a308e50ff6d00ff2375275d81ca4f9c6fcf1
INFO:victron_ble.scanner:1736142950.0540771      : 100038a308e60ff681afbf623fe12d64600cbdd9d9
INFO:victron_ble.scanner:1736142950.539347       : 100038a308e70ff6f751996b8dd1534c7f5877f170
INFO:victron_ble.scanner:1736142951.9574435      : 100038a308e80ff625625f3dd3925152c3c9da7dc8
INFO:victron_ble.scanner:1736142952.9031525      : 100038a308e90ff69ec17cd31772d84b9765bdf756

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Shane11r, 2025-01-06:
for what its worth i copied the solar charger template and found most of the results came back accurate with the exception of current. while current should have being reading 0 that template returns -25.6
{"name": "BSC IP65 12/10 HQ2341NQ7PG", "address": "E7:94:5B:3D:F4:53", "rssi": -66, "payload": {"battery_charging_current": -25.6, "battery_voltage": 12.8, "charge_state": "power_supply", "charger_error": "no_error", "model_name": "Blue Smart Charger BSCIP65 12|10 Battery Charger"}}

Magic01, 2025-02-15:
I also like to see this. Is the project still alive?

keshavdv, 2025-05-05:
This should be supported in the latest release

closed 2025-05-05 by keshavdv